strands-project/scitos_apps #128: [scitos_docking] How to configure injected pose

hawesie, 2015-01-27:
How do we configure the pose which is injected after successful docking? We have a map which we wish to reuse but with the docking station not at the origin.

hawesie, 2015-02-25:
@gestom any news on this?

gestom, 2015-03-18:
Should be solved now. To determine the position of the charging station in the map, dock to it (automatically) and ask for the /robot_pose. Note that the station and robot orietation oppose each other.

hawesie, 2015-03-18:
> Note that the station and robot orietation oppose each other.

Does that mean the orientation from `/robot_pose` should be rotated 180 before being used for the parameter for the station?

gestom, 2015-03-18:
yeap, exactly, stationPhi = robot_pose.yaw+\pi

marc-hanheide, 2015-03-19:
it's now released, so I close this.

closed 2015-03-19 by marc-hanheide