strands-project/scitos_apps #134: [scitos_docking] fails to build on jenkins

cdondrup, 2015-03-19:
Now in the correct repo. Sorry. This is the output:
```
[ 97%] Building CXX object CMakeFiles/visual_charging.dir/src/scitos_docking/docking.cpp.o
/usr/bin/x86_64-linux-gnu-g++   -DROSCONSOLE_BACKEND_LOG4CXX -DROS_PACKAGE_NAME=\"scitos_docking\" -g -O2 -fstack-protector --param=ssp-buffer-size=4 -Wformat -Werror=format-security -D_FORTIFY_SOURCE=2  -I/tmp/buildd/ros-indigo-scitos-docking-0.0.15-0trusty-20150319-1122/obj-x86_64-linux-gnu/devel/include -I/tmp/buildd/ros-indigo-scitos-docking-0.0.15-0trusty-20150319-1122/include -I/opt/ros/indigo/include    -o CMakeFiles/visual_charging.dir/src/scitos_docking/docking.cpp.o -c /tmp/buildd/ros-indigo-scitos-docking-0.0.15-0trusty-20150319-1122/src/scitos_docking/docking.cpp
In file included from /opt/ros/indigo/include/ros/ros.h:40:0,
                 from /tmp/buildd/ros-indigo-scitos-docking-0.0.15-0trusty-20150319-1122/include/scitos_docking/CTransformation.h:15,
                 from /tmp/buildd/ros-indigo-scitos-docking-0.0.15-0trusty-20150319-1122/src/scitos_docking/docking.cpp:5:
/tmp/buildd/ros-indigo-scitos-docking-0.0.15-0trusty-20150319-1122/src/scitos_docking/docking.cpp: In function 'void mainLoop()':
/opt/ros/indigo/include/ros/console.h:342:176: error: format not a string literal and no format arguments [-Werror=format-security]
     ::ros::console::print(filter, __rosconsole_define_location__loc.logger_, __rosconsole_define_location__loc.level_, __FILE__, __LINE__, __ROSCONSOLE_FUNCTION__, __VA_ARGS__)
                                                                                                                                                                                ^
/opt/ros/indigo/include/ros/console.h:345:5: note: in expansion of macro 'ROSCONSOLE_PRINT_AT_LOCATION_WITH_FILTER'
     ROSCONSOLE_PRINT_AT_LOCATION_WITH_FILTER(0, __VA_ARGS__)
     ^
/opt/ros/indigo/include/ros/console.h:375:7: note: in expansion of macro 'ROSCONSOLE_PRINT_AT_LOCATION'
       ROSCONSOLE_PRINT_AT_LOCATION(__VA_ARGS__); \
       ^
/opt/ros/indigo/include/ros/console.h:516:35: note: in expansion of macro 'ROS_LOG_COND'
 #define ROS_LOG(level, name, ...) ROS_LOG_COND(true, level, name, __VA_ARGS__)
                                   ^
/opt/ros/indigo/include/rosconsole/macros_generated.h:98:23: note: in expansion of macro 'ROS_LOG'
 #define ROS_INFO(...) ROS_LOG(::ros::console::levels::Info, ROSCONSOLE_DEFAULT_NAME, __VA_ARGS__)
                       ^
/tmp/buildd/ros-indigo-scitos-docking-0.0.15-0trusty-20150319-1122/src/scitos_docking/docking.cpp:710:4: note: in expansion of macro 'ROS_INFO'
    ROS_INFO(status);
    ^
cc1plus: some warnings being treated as errors
```

marc-hanheide, 2015-03-19:
https://github.com/strands-project/scitos_apps/pull/133 should fix it.

closed 2015-03-19 by marc-hanheide

marc-hanheide, 2015-03-19:
built now!