strands-project/scitos_apps #137: [docking] less accuracy since firmware update

bfalacerda, 2015-03-30:
Since we did the firmware update before the marathon, it looks like docking is failing a lot more. I guess it has something to do with the softer breaking with the new firmware.

Is anyone else experiencing this?

nilsbore, 2015-03-30:
We haven't really tested docking with new drivers yet since we've been using the old ones. I'll run a few tests tomorrow.

nilsbore, 2015-03-31:
I ran it a couple of times and needed some retries before succeeding. This is more or less consistent with what I've observed before. Though it seems like it's driving with a slightly lower force than it did before. Maybe that can cause it to not make it all the way up the charger?

gestom, 2015-03-31:
I will have a look at it. Could you send me some statistics about the failures (let's say from 100 docking attempts - both before and after the firmware upgrade) + logs from the charging action server's feedback ?

bfalacerda, 2015-03-31:
is this feedback saved anywhere?

gestom, 2015-04-01:
It should be in the logs.

bfalacerda, 2015-04-01:
?
Please be more specific. If you mean the normal ROS logs, we dont save them, I doubt we'll have any data from pre firmware update

gestom, 2015-04-01:
The docking station's action server feedback is also reported through ROS_INFO and ROS_DEBUG, which goes in the ROS logs, no?

gestom, 2015-04-01:
Anyway, I will be perfectly willing to have a look at the issue, because the charging is quite important for the deployment, but solving that makes sense only if both AAF and G4S robots have the same firmware version. Changing the charging code to conform with the new firmware could actually screw the AAF robot's self-recharging ability. Anyway, can you quantify the percentage of unsuccessfull dokings, retries, average times etc. and report what exactly differs from before the upgrade ?

bfalacerda, 2015-04-01:
Here's a link to the .ros folder in bob. I dont think we have anything else, this should be just some post  firmware update data.

https://www.dropbox.com/s/o0cismo3bz6yqmr/ros_logs.tar.gz?dl=0

I can't really quantify anything else, as I said it *looks* like docking is failing more, but this just comes from me noticing fails more often.

I wasn't aware that we were not using the latest firmware for AAF

bfalacerda, 2015-04-07:
after some more testing today, I think this was mainly caused by the rest of the navigation pipeline not working properly, and not due to lack of accuracy on docking itself, so I'll close

"Fixed" by https://github.com/strands-project/strands_navigation/pull/216

closed 2015-04-07 by bfalacerda