strands-project/scitos_apps #140: [scitos_docking] wasting a lot of CPU time

marc-hanheide, 2015-05-21:
because `scitos_docking` is always subscribed to many topics (https://github.com/strands-project/scitos_apps/blob/hydro-devel/scitos_docking/src/scitos_docking/docking.cpp#L745-L754) and actually even does quite some processing in the callbacks (e.g. https://github.com/strands-project/scitos_apps/blob/hydro-devel/scitos_docking/src/scitos_docking/docking.cpp#L268 and https://github.com/strands-project/scitos_apps/blob/hydro-devel/scitos_docking/src/scitos_docking/docking.cpp#L215) it keeps the CPU very busy all the time. This is very bad, in particular as CPU is a scarce resource on the main PC. It is suggested that visual docking unsubscribes from all topics it does not currently need, and only subscribes when it is active.

I'm happy to help with that if need be.

closed 2015-05-22 by marc-hanheide

gestom, 2015-05-26:
The docking is not supposed to do any image processing if in the IDLE state:https://github.com/strands-project/scitos_apps/blob/hydro-devel/scitos_docking/src/scitos_docking/docking.cpp#L306. Moreover, the main loop has a 30 ms sleep inside, so the callbacks cannot be triggered that often. As far as I remember, I have checked the CPU whenever making changes to it and it was always below 1% unless it was activated. Having a full CPU load means that 1) something is making the docking server active 2) it is related to some recent changes. My guess is that some component calls charging by mistate with a very long timeout.

marc-hanheide, 2015-05-26:
It was in fact not using as much CPU as when it was active. There were just several threads, each consuming about 12-20%. I'd blame this on image transport that processes the images internally. Also it was causing a lot of unnecessary network traffic. I say one should also unsubscribe from everything that's not needed whenever possible to minimise the ram and CPU footprint.

gestom, 2015-05-26:
OK

PR #141 (2015-05-22): (un)subscribe from all the topics
this is to fix #140 by unsubscribing from all the topics when we don't have a task to charge or calibrate. To be checked on the robot by @Jailander. Also, @gestom, please confirm this is acceptable.