strands-project/scitos_apps #146: using scitos_docking with openni2

ToMadoRe, 2015-08-05:
Using openni2, the topic head_xtion/image/mono didn't work for me, but ../image/raw fixed it. 
https://github.com/strands-project/scitos_apps/blob/hydro-devel/scitos_docking/src/scitos_docking/docking.cpp#L760

cburbridge, 2015-08-05:
On Bob I'm using the latest released version of scitos_docking, and it works fine with openni2_launch...

ToMadoRe, 2015-08-05:
The topic is listed, but `rostopic hz head_xtion/rgb/image_mono` doesn't return anything.

cburbridge, 2015-08-06:
Did you launch openni as in https://github.com/strands-project/strands_systems/issues/113 ? Could it be related to having debayer_processing:=true?

ToMadoRe, 2015-08-06:
no, I didn't. Maybe that's why. I just started it with 'roslaunch openni2_launch openni2.launch depth_registration:=true'

cburbridge, 2015-08-06:
I think we should try to fix things as you did to avoid having to start with debayering, since it should not be needed. At the review we put it on to get some extra topics for some component, but I don't think it was charging..

cdondrup, 2015-08-06:
@gestom is currently on holiday so there is no way of knowing if using a colour image is somehow detrimental to the circle detector. I'll ping him when he is back.

bfalacerda, 2016-09-30:
I'm reviving this issue because I just had the same problem trying to dock and failing, and wasnt aware that we needed to run with ``debayer_processing:=true``. Is this really needed, or can we change  docking to look at ``head_xtion/rgb/image_raw `` @gestom ?