strands-project/scitos_apps #152: scitos_docking with two stations

hawesie, 2016-04-15:
For our TSC deployment we are going to have two docking stations. Does this work with the calibration or does the node need updating to distinguish between multiple docking stations? @Jailander @gestom

Jailander, 2016-04-15:
Good question, I don't think this has been implemented yet, I can take a look to make it possible unless @gestom see this before I take a look and comes with a suggestion, it might mean running two instances of scitos visual docking though. 
*Also* apart from the calibration you also need these two things to consider:
* Make sure that scitos_docking is launched with the **positionUpdate** parameter set to **False** as it is [here](https://github.com/strands-project/strands_systems/blob/indigo-devel/strands_bringup/launch/strands_navigation.launch#L28), and  
* In the topological map the [localise_anywhere](https://github.com/strands-project/strands_navigation/tree/indigo-devel/topological_navigation#using-localise-by-topic) field for the Charging Stations should be set to **False**. I can of course do this.

hawesie, 2016-04-15:
The two instances make sense as long as everything can be namespaced properly. Good idea.

Jailander, 2016-04-17:
@hawesie after taking a look at the code, I think the easiest and best solution would be to make sure that the calibration works with both charging stations, this shouldn't be too hard as long as there aren't big differences in relative positioning of the label and the station between both stations. Do you think this is feasible?

If not namespacing everything is also possible but it would be hassle and some things like the docking/undocking using the joystick button need some redesign.

hawesie, 2016-04-17:
So pretend it's the same station twice? That feels risky, but if you think it will be ok, then we can try it. I would like to test all this in simulation this week, but I can't get the docking to work in simuation: https://github.com/strands-project/strands_morse/issues/158. Any chance you could take a look at this and then we can work together on the double charging stations?

Jailander, 2016-04-17:
Yes I have tried it in Linda today and it works reasonably ok, it will need more testing though, in simulation I saw the issue will take a look at it tomorrow morning but I think could be related to some difference between the cameras parameters in simultation and in the robots.

I'll try to identify which ones are the problematic ones

hawesie, 2016-04-18:
Great, thanks.

hawesie, 2016-04-18:
Having seen the simulation PR, I'm more nervous. How exact do the similarities need to be for this to work? Can we make that work in reality?

Jailander, 2016-04-18:
Not very can't give you an exact number, you have enough leg room. I have tested it using two labels places at the same distance from the floor and moving the charging station from one to the other label placing it by hand (carefully but still by hand).

My only concern is that in Linda we never installed the rubber protections over the charging station plates in this case it could not work.

Jailander, 2016-04-19:
Hi @hawesie I have done this test to test how well scitos docking would work on this situation:
* I placed the robot manually as centred as possible on the charging station with the camera facing the wall.
* then I placed the label in the centre of the image and calibrated.
* I tried docking and undocking moving the label moving the label +/- 5cm on both axis 20 times and it was successful
My only concern would be the rubber patches you have on bob's charging station could you do some testing in Bham if not I could do it on Friday when I go.

gestom, 2016-04-19:
Hi, sorry for the late reply, but I was off-the-grid the last days. Since we have a new version of the circle detector, which can encode IDs in the circles, we could simply encode the station ID in the physical tag. The charging would then read the appropriate station ID when (un)-docking and read station-specific parameters.

Jailander, 2016-04-19:
Yes I thought so, but didn't want to commit to do it without you, I think for now we can use this not ideal solution and when you are back (which will be before the deployment anyway @hawesie ) we can work on it.

hawesie, 2016-04-19:
We will install the nav stuff from May 2nd I guess. Can you do it by then? If so, awesome.

gestom, 2016-05-07:
Unfortunately, encoding the ID in the circles work only on short distances and is kind of ugly. I tried a different scheme and it seems that we could distinguish between four stations: https://github.com/gestom/scitos_apps/tree/hydro-devel/scitos_docking/doc.  I wil try to finish and test it by today.

gestom, 2016-05-07:
Seems to work OK: https://github.com/strands-project/scitos_apps/pull/156