strands-project/scitos_apps #153: very low speeds with undocking

bfalacerda, 2016-04-28:
I have calibrated the docking, and betty docks normally. However, when i tell her to undock, it seems to start with normal negative speeds, but then it gets veryc close to zero and the robot doesn't move anymore. here is the firt few published velocities:

```
linear: 
  x: -0.5
  y: 0.0
  z: 0.0
angular: 
  x: 0.0
  y: 0.0
  z: 0.0
---
linear: 
  x: -0.5
  y: 0.0
  z: 0.0
angular: 
  x: 0.0
  y: 0.0
  z: 0.0
---
linear: 
  x: -0.5
  y: 0.0
  z: 0.0
angular: 
  x: 0.0
  y: 0.0
  z: 0.0
---
linear: 
  x: -0.5
  y: 0.0
  z: 0.0
angular: 
  x: 0.0
  y: 0.0
  z: 0.0
---
linear: 
  x: -0.00750212669373
  y: 0.0
  z: 0.0
angular: 
  x: 0.0
  y: 0.0
  z: 0.0
---
linear: 
  x: -0.00750212669373
  y: 0.0
  z: 0.0
angular: 
  x: 0.0
  y: 0.0
  z: 0.0
---
linear: 
  x: -0.00750212669373
  y: 0.0
  z: 0.0
angular: 
  x: 0.0
  y: 0.0
  z: 0.0
---
linear: 
  x: -0.00750212669373
  y: 0.0
  z: 0.0
angular: 
  x: 0.0
  y: 0.0
  z: 0.0
---
linear: 
  x: -0.00750212669373
  y: 0.0
  z: 0.0
angular: 
  x: 0.0
  y: 0.0
  z: 0.0
```

It keeps publishing these very slow speeds afterwards.

cdondrup, 2016-04-28:
Maybe the obstacle detection?

gestom, 2016-04-28:
Probably obstacle detection as pointe out by @cdondrup . Can you send a rosbag with the head camera topics ?

Jailander, 2016-04-28:
I think Betty might be having issues with the canera topics because of this and some other issues with the calibration but I guess @gestom will know better

bfalacerda, 2016-04-28:
will send the rosbags tmrw. She's able to dock though. Is there any calibration we can run for the head cam? We did have to repare her head cam halo, so it might not be exactly the same as before

bfalacerda, 2016-04-29:
@gestom here is a successful dock and an unsuccessful undock:
https://www.dropbox.com/s/wi0mzp4z2mcj3j6/dock.zip?dl=0
https://www.dropbox.com/s/qsi2lqs2nt92rei/undock.zip?dl=0

bfalacerda, 2016-04-29:
also, there are some network issues and the rate of the  head cam on betty is reduced:

```
strands@betty:/media/strands/PLANSIG2013/bham_integration/semantic_maps$ rostopic hz /head_xtion/rgb/image_raw
sh: 0: getcwd() failed: No such file or directory
subscribed to [/head_xtion/rgb/image_raw]
average rate: 12.678
	min: 0.078s max: 0.080s std dev: 0.00044s window: 12
average rate: 12.670
	min: 0.078s max: 0.080s std dev: 0.00053s window: 24
average rate: 12.669
	min: 0.078s max: 0.080s std dev: 0.00052s window: 37
average rate: 12.669
	min: 0.078s max: 0.080s std dev: 0.00052s window: 50
^Caverage rate: 12.670
```

gestom, 2016-04-29:
@bfalacerda:  My zip version complaints that the files are incomplete, can you check ?

bfalacerda, 2016-04-29:
yeah, there was something wrong with them. Please try again

gestom, 2016-04-29:
Originally, the depth image data encoding was UINT16, now it's FLOAT32. This was not reflected when calculating obstacle distance during undocking.

cdondrup, 2016-04-30:
It still works on Linda using the new openni2 launch files though. Are we sure this is not just a Betty issue?

closed 2016-05-03 by bfalacerda