strands-project/scitos_apps #166: Split repo into STRANDS vs scitos generic

hawesie, 2021-08-04:
I am in the process of setting up parts of the scitos stack on a scitos X3. This doesn't have all the things the STRANDS Scitos A5/G5 has, therefore I only need some of the packages from this repo. This led me to consider that we should split this into `scitos_apps` that work for all scitos robots, and `strands_scitos_apps` that are specific to the STRANDS robots. I think the generic parts would be at least:

scitos_cmd_vel_mux
scitos_dashboard
scitos_teleop

What do our users think? (probably just @marc-hanheide)

marc-hanheide, 2021-08-05:
Thanks @hawesie did you consider moving those packages into `scitos_drivers` repo rather than creating yet another repo? To me, that would probably make most sense.

hawesie, 2021-08-05:
That's a fantastic idea. Shall I go ahead and do that?

marc-hanheide, 2021-08-05:
Please do! Remember to ideally transfer with git history and updating the version. Indeed, the new version of all packages in the `scitos_drivers` must be the larger of the two sets of packages going in there, to ensure the update processes for packages all work. Bit annoying, but shouldn’t be a problem.

hawesie, 2021-08-05:
Gotcha.